PumucklOnTheAir/TestFramework #3: Testframework Basis

P8H, 2015-11-09:
Basis anlegen Threading für Tests

P8H, 2015-11-09:
womöglich interessant für dich 
https://docs.python.org/2/library/ipc.html

P8H, 2015-12-15:
Prozesse nutzen statt Threads, wegen VLAN

P8H, 2016-01-11:
tests schlagen auf dem raspberry pi fehl

Traceback (most recent call last):
  File "/usr/lib/python3.4/concurrent/futures/_base.py", line 297, in _invoke_callbacks
    callback(self)
  File "/home/pi/TestFramework_1/server/server.py", line 193, in _test_done
    cls.__start_task(task.router, None)
  File "/home/pi/TestFramework_1/server/server.py", line 132, in __start_task
    task = cls.executor.submit(cls._execute_test, job, router)
  File "/usr/lib/python3.4/concurrent/futures/process.py", line 396, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown

P8H, 2016-01-11:
python3.4 -m unittest framework_unittests/test_Server_2.py

P8H, 2016-01-11:
Ordner TestFramework_1 auf dem Pi

Symen, 2016-01-20:
NVAssistent.create_namespace_vlan_veth(router) sollte jetzt verwendet werden, um in einem neuen Prozess VLANs zu nutzen.

P8H, 2016-01-25:
NVAssistent.create_namespace_vlan_veth(router) sollte wohl eher ein _RemoteSystem_ als parameter haben oder @Symen ?

P8H, 2016-01-25:
und zum schluss nv_assi.close() aufrufen? @Symen

closed 2016-02-24 by P8H