PumucklOnTheAir/TestFramework #51: Error nach Start des Servers

Symen, 2015-12-16:
root@bunsenlabs:~/TestFramework# python3.4 start_server.py 
2015-12-16 16:59:02,295 - INFO     : 	Create Network Controller for Router(21) ...
2015-12-16 16:59:02,685 - DEBUG    : 		Create VLAN Interface ...
2015-12-16 16:59:03,332 - DEBUG    : 			Wait for ip assignment via dhcp for VLAN Interface(vlan21) ...
2015-12-16 16:59:05,845 - DEBUG    : 			[+] vlan21 created with: Link=eth0, VLAN_ID=21, IP=192.168.1.215/24
2015-12-16 16:59:05,846 - DEBUG    : 		Create Namespace ...
2015-12-16 16:59:06,037 - DEBUG    : 			[+] Namespace(nsp21) successfully created
2015-12-16 16:59:06,689 - DEBUG    : 			[+] Encapsulate Interface(vlan21)
2015-12-16 16:59:06,691 - INFO     : 	Connect with Router(21) ...
2015-12-16 16:59:08,116 - DEBUG    : 		[+] Successfully connected with Router(21)
2015-12-16 16:59:08,126 - DEBUG    : 		[+] Sent the command (cat /proc/cpuinfo | grep machine) to the Router(21)
2015-12-16 16:59:08,191 - DEBUG    : 		[+] Sent the command (uci show network.client.macaddr) to the Router(21)
2015-12-16 16:59:08,260 - DEBUG    : 		[+] Sent the command (uci show wireless.client_radio0.ssid) to the Router(21)
2015-12-16 16:59:08,275 - INFO     : 	Disconnect with Router(21) ...
2015-12-16 16:59:08,278 - DEBUG    : 		Delete VLAN Interface ...
2015-12-16 16:59:08,282 - DEBUG    : 			[+] Interface(vlan21) is already deleted
2015-12-16 16:59:08,283 - DEBUG    : 		Delete Namespace ...
2015-12-16 16:59:08,362 - DEBUG    : 			[+] Namespace(nsp21) successfully deleted
Runtime Server started
Traceback (most recent call last):
  File "start_server.py", line 4, in <module>
    Server.start()
  File "/root/TestFramework/server/server.py", line 47, in start
    cls._ipc_server.start_ipc_server(cls, True)  # serves forever - works like a while(true)
  File "/root/TestFramework/server/ipc.py", line 26, in start_ipc_server
    self._server_object = self.get_server()
  File "/usr/lib/python3.4/multiprocessing/managers.py", line 460, in get_server
    self._authkey, self._serializer)
  File "/usr/lib/python3.4/multiprocessing/managers.py", line 140, in __init__
    self.listener = Listener(address=address, backlog=16)
  File "/usr/lib/python3.4/multiprocessing/connection.py", line 447, in __init__
    self._listener = SocketListener(address, family, backlog)
  File "/usr/lib/python3.4/multiprocessing/connection.py", line 584, in __init__
    self._socket.bind(address)
OSError: [Errno 99] Cannot assign requested address

P8H, 2015-12-16:
läuft der Server womöglich schon im Hintergrund?
Auf dem Raspberry Pi befindet er sich im autostart modus. 
_systemctl status fftserver_
könnte weiterhelfen.
Siehe README.

Symen, 2015-12-16:
ne der Server läuft nicht im Hintergrund.  Habs auch nur auf meinem laptop ausprobiert

P8H, 2015-12-16:
fehler tritt in welcher Branch auf?

P8H, 2015-12-16:
branch firmware-#33 ausprobiert. Fehler bisher nicht reproduzierbar.

Bekannter Fehler wen der Server doppelt gestartet wird:
OSError: [Errno 48] Address already in use

P8H, 2015-12-16:
Schon mal Rechner neugestartet?  Vllt konnte er durch einen Abbruch des Servers den Port nicht mehr freischalten.

Hab trotzdem mal bei IPC '127.0.0.1' auf 'localhost' geändert. Allerdings hat es ja früher doch auch immer funktioniert?

Symen, 2015-12-16:
Ja hatte  einen Neustart gemacht und den gleichen Fehler bekommen.
Werds morgen nochmal auf dem pi versuchen.

Kannst dir ja auch nochmal angucken was ich in server.py hizugefügt hab.
Gibt jetzt drei neue Funktionen.

Symen, 2015-12-28:
ja solange networkctrl in einem eigenen Thread läuft funktioniert alles

closed 2015-12-28 by Symen