PumucklOnTheAir/TestFramework #54: (22) Erweiterung Firmwareveraltung

P8H, 2015-12-17:


Symen, 2015-12-20:
1. Firmwares können jetzt einzeln oder alle auf einmal runtergeladen werden.
2. Bereits runtergeladenen Firmwares können importiert werden.
4. Ein Webserver wird mit dem NetworkCtrl gestartet
3. Der Router downloaded sich die Firmware vom Webserver per wget-Kommando

P8H, 2015-12-20:
enthält auch commits aus cli #46. @acid-horny 
Falls der Stand stable ist, kann ein Pull Request erstellt werden.

Dann werden euch auch direkt die Fehler angezeigt durch travis.
Alternativ auf https://travis-ci.org/PumucklOnTheAir/TestFramework/branches gucken.

Hat aber alles keine Eile ;)

P8H, 2015-12-27:
hab mit Lena nochmal gesprochen, anscheinend ist die Branch schon bereit.

Kannst also ein Pull Request erstellen. :+1:

Symen, 2015-12-28:
Pull Request ist erstellt. Allerdings kam folgende Meldung:

"This branch has conflicts that must be resolved
Use the command line to resolve conflicts before continuing. "

Hab jetzt mal gemacht was da stand. Der master-branch bei mir aufm Rechner sollte auf dem neuesten Stand sein, hab ihn aber noch nicht gepusht. War irgendwie nur die README.md mit einem Konflikt.

Normal war doch nichts über die "command line" zu machen, oder?

P8H, 2015-12-28:
Normalerweise nicht, kannst du daraus einfach eine neue Branch ableiten und daraus ein PR erstellen?

Symen, 2015-12-28:
So glaub jetzt stimmt alles. Habs nochmal anders versucht.

closed 2015-12-29 by P8H