PumucklOnTheAir/TestFramework #75: test_firmware_handler.py overtime

P8H, 2016-01-13:
tests/test_firmware_handler.py benötigt ziemlich viel Zeit (manchmal mehr als 2 Min) auf Travis CI.

Muss das sein?
Man kann auf travis z.B. einstellen das bestimmte Ordner/Dateien gecacht werden. Würde das Helfen oder den Test nutzlos machen?

P8H, 2016-01-13:
es liegt am neuem py.test..
https://travis-ci.org/PumucklOnTheAir/TestFramework/builds/102102326
https://travis-ci.org/PumucklOnTheAir/TestFramework/builds/101617549

oder dass der File Server uns geblockt hat :))

P8H, 2016-01-14:
lag wohl doch eher am coverage test tool

Symen, 2016-01-14:
also test_firmware_handler.py kann durchaus lange dauern, kommt drauf an ob alle "Images" runtergeladen werden sollen.

Symen, 2016-01-14:
der Wert download_all sollte eigentlich aber auf False gesetz sein, so dass nur ein "Image" runtergeladen wird

closed 2016-02-23 by P8H