PumucklOnTheAir/TestFramework #107: Error: Can't get attribute 'RouterInfoJob'

Symen, 2016-02-10:
Wird ein job ausgeführt und danach ein weiterer (kann auch nochmal der gleiche sein) bekommt man einen Error:

2016-02-10 16:33:11,631 - DEBUG    : 	Starting task RouterInfoJob
Process ForkPoolWorker-5:
2016-02-10 16:33:11,649 - DEBUG    : waiting......
Traceback (most recent call last):
  File "/usr/lib/python3.4/multiprocessing/process.py", line 254, in _bootstrap
    self.run()
  File "/usr/lib/python3.4/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.4/multiprocessing/pool.py", line 108, in worker
    task = get()
  File "/usr/lib/python3.4/multiprocessing/queues.py", line 357, in get
    return ForkingPickler.loads(res)
AttributeError: Can't get attribute 'RouterInfoJob' on <module 'util.router_info' from '/home/pi/TestFramework_simon/util/router_info.py'>

Symen, 2016-02-10:
Um den Fehler zu reproduzieren:
1. Server starten (Anzahl Router ist egal)
2. python3 cli.py update_info -r 0
3. python3 cli.py update_info -r 0

P8H, 2016-02-22:
in der Branch testbasic-#3 nicht reproduzierbar.