PumucklOnTheAir/TestFramework #108: RemoteSystemJob: Warum Thread?

Symen, 2016-02-15:
Warum soll "RemoteSystemJob" eigentlich ein Thread sein?
Ein erbendes Objekt wie zum Beispiel "RouterOnlineJob(RemoteSystemJob)" ruft ja dann sowie so nochmal einen eigenen Thread auf. Außerdem müsste Thread.\__init__() in "RemoteSystemJob" aufgerufen werden, was allerdings Probleme gibt da das Objekt dann anscheinend nicht mehr so ohne weiteres an einen neuen Prozess übergeben werden kann. 

Ich denk es ist besser, wenn "RemoteSystemJob" kein Thread ist. Funktioniert dann auf jeden Fall besser. Vorher konnte ich den Server nicht ohne Fehler starten.

In der Server Klasse müsste dann nur folgendes geändert werden:
```
@classmethod
    def _execute_job(cls, job: RemoteSystemJob, remote_sys: RemoteSystem, data: {}) -> {}:
        # proofed: this method runs in other process as the server
        Logger().debug("Execute job " + str(job) + " on " + str(remote_sys), 2)
        #Alt:
        #Thread.__init__(job)
        job.prepare(remote_sys, data)

        cls.__setns(remote_sys)

        #Alt: remote_system kein Thread mehr
        #job.start()
        #job.join(300)  # Timeout: 5 minutes
        #Neu:
        try:
            job.run()
        except Exception:
            Logger().debug("Error while exectue job " + str(job))
        result = job.get_return_data()

        return result
```

P8H, 2016-02-15:
das hat historische Gründe. Mit dem altem System ging es gar nicht anders. Ja, man könnte nun rausnehmen. 

(Thread.init() wird/muss immer im neuem Prozess gestartet werden)

Ich überprüfe es mal...

P8H, 2016-02-22:
Job nun ohne Threads. Umgesetzt in testbasic-#3

closed 2016-02-22 by P8H