PumucklOnTheAir/TestFramework #131: unstable ssh connections

P8H, 2016-03-07:
Beobachtung:
SSH Session sind sehr langsam und hängen sich auf. Das selbe gilt für PyCharm Remote Prozeduren.

Bisherige Analyse:

P8H, 2016-03-07:
im commit #be6f1d4 existiert der Bug auch, allerdings nur zur init Zeit. Vermutlich besteht ein Zusammenhang mit den Namespace/VLAN da sie in dieser Version jedesmal abgebaut werden nach einem ausgeführtem Job.

P8H, 2016-03-07:
im aktuellem master (8d1c52e) funktioniert es die ganze Zeit nicht, so lange der Server läuft. Nach dem beenden des Servers, ist der ping wieder stabil.

P8H, 2016-03-07:
habe ein Namespace erstellt nach der Anleitung im Wiki.
(https://github.com/PumucklOnTheAir/TestFramework/wiki/Create-VLan-and-Namespace-by-console)

Allerdings konnte das nicht den Fehler reproduzieren.

Symen, 2016-03-10:
Bei mir funktionieren alle Funktionen im Master bzw. hab update_info, online, reboot, ausprobiert und keine Errors bekommen. Router wurden auch korrekt neugestartet.

P8H, 2016-03-10:
Fehler unabhängig von den Funktionen vom Framework. Framework muss bloß gestartet sein, um die Symptome zu erzeugen.

P8H, 2016-03-10:
nicht mehr reproduziebar..

closed 2016-03-10 by P8H