PumucklOnTheAir/TestFramework #154: Prozesse werden nicht geschlossen

Symen, 2016-03-21:
Sobald ein Task, sei es jetzt Job oder Test, ausgeführt wird, bleibt dieser auch nach beenden offen. Sprich wir bekommen immer mehr offene Prozesse. Kann man sich mit "pstree" angucken. Wird eine ganze Liste von "python3" Prozessen.

P8H, 2016-03-21:
Branch task-bug-#154

Prozess Title hinzugefügt.
mit pstree sieht man nun dass die richtige Anzahl der Prozesse erstellt und wieder geschlossen wird. Auch das "recycling" der Prozesse durch das Pool Objekt, findet immer direkt nach beenden eines Prozesses statt.

P8H, 2016-03-21:
Die steigende Anzahl an Prozesse könnten aus dem IPC Manager z.B. stammen oder andere Module die im Server eingebunden sind.

closed 2016-03-21 by P8H