PumucklOnTheAir/TestFramework #186: Frage zum Logger

Symen, 2016-03-28:
@Sniger87 mal ne Frage: 
Warum gibt folgender Code, wenn ich ihn ganz normal ausführe nicht auch 
"[+] done with work" 
zurück?
```
import logging
from log.loggersetup import LoggerSetup
from time import sleep

class TestLogger:

    def __init__(self):
        LoggerSetup.setup()

    def no_subprocess(self):
        logging.info("%stest_no_subprocess ", LoggerSetup.get_log_deep(1))
        logging.debug("%sdo some work ...", LoggerSetup.get_log_deep(2))
        sleep(5)
        logging.debug("%s[+] done with work", LoggerSetup.get_log_deep(2))

    def close(self):
        LoggerSetup.shutdown()

tl = TestLogger()
tl.no_subprocess()
tl.close()
```

Ergebnis im log-file:
```
2016-03-28 11:13:00,009 - INFO     : 	test_no_subprocess 
2016-03-28 11:13:00,014 - DEBUG    : 		do some work ...
```

Sniger87, 2016-03-28:
es kann gut möglich sein das %s[ nicht als platzhalter interpretiert wird.
versuch mal %s [ mit einem leerzeichen dazwischen, ob dies funktioniert

Symen, 2016-03-28:
Ne daran liegt es nicht.
Wenn ich folgenden Code ausführe bekomme ich nämlich die Zeile angezeigt.

```
...
for i in range(0, 2):
    tl = TestLogger()
    tl.no_subprocess()
    tl.close()
```
Ergebnis:
```
2016-03-28 12:02:46,126 - INFO     : 	test_no_subprocess 
2016-03-28 12:02:46,130 - DEBUG    : 		do some work ...
2016-03-28 12:02:51,134 - DEBUG    : 		[+] done with work
2016-03-28 12:02:51,134 - INFO     : 	end test
2016-03-28 12:02:51,146 - INFO     : 	test_no_subprocess 
2016-03-28 12:02:51,149 - DEBUG    : 		do some work ...
```
Interessant ist auch, dass das Ergebnis nicht immer gleich ist. Ich hatte auch schon:
```
2016-03-28 12:02:46,126 - INFO     : 	test_no_subprocess 
2016-03-28 12:02:46,130 - DEBUG    : 		do some work ...
2016-03-28 12:02:51,134 - DEBUG    : 		[+] done with work
2016-03-28 12:02:51,134 - INFO     : 	end test
2016-03-28 12:02:51,146 - INFO     : 	test_no_subprocess 
2016-03-28 12:02:51,149 - DEBUG    : 		do some work ...
2016-03-28 12:02:51,150 - DEBUG    : 		[+] done with work
2016-03-28 12:02:51,152 - INFO     : 	end test
```

Symen, 2016-03-28:
Hab das Gefühl irgendwas wird zu früh geschlossen.
Mit sleep(1) am Ende, bleibt die Ausgabe anscheinend konstant richtig.

```
...
def no_subprocess(self):
        logging.info("%stest_no_subprocess ", LoggerSetup.get_log_deep(1))
        logging.debug("%sdo some work ...", LoggerSetup.get_log_deep(2))
        sleep(5)
        logging.debug("%s[+] done with work", LoggerSetup.get_log_deep(2))
        logging.info("%send test", LoggerSetup.get_log_deep(1))
        sleep(1)
...
```

Symen, 2016-03-28:
Und noch was interessantes ;)
Wenn ich das ganze auf der Konsole mit root-Rechten ausführe bekomme ich nichts angezeigt. Ohne root-Rechte allerdings schon, obwohl beide die gleiche Konsole nutzen (/dev/pts/2).

Symen, 2016-03-28:
Wo wird denn "_stream_paths" aus dem LoggerSetup verwendet? Seh nur dass immer was mit append hinzugefügt wird nicht aber genutzt.

Sniger87, 2016-03-28:
ja sieht fast so aus, als ob handler zu früh geschlossen und verworfen werden oder sogar die logging-Instanz an sich selbst

closed 2016-03-30 by Sniger87