PumucklOnTheAir/TestFramework #188: Keine Ausgabe der SubProzesse auf der CLI

Symen, 2016-03-28:
Führt man z.B. 

`python3 cli.py online -r 0`

aus, wird auf der Konsole der CLI nur das ausgegeben was vom Server-Prozess kommt. Erst nach mehrmaligem aufrufen des genannten Befehls werden auch Ausgaben der Child-Prozesse angezeigt.

Symen, 2016-03-28:
Lässt man sich in einer run-Funktion einer der util-Klassen die Handler des Loggers ausgeben kann man sehen, dass ein Handler fehlt. Bzw. sobald auf der CLI-Konsole die richtigen Ausgaben sichtbar sind gibt es einen Handler mehr.

Symen, 2016-03-28:
Schreibt man die gewünschte Konsole direkt ins LoggerSetup funktioniert es. Man könnte also zu Beginn auch alle Konsolen bestimmen die aktive sind (im terminal z.B. mit 'who'. Geht in python sicher auch irgendwie) und diese dann direkt hinzufügen. Vorteil wäre die CLI müsste nicht vor jedem Befehl die Konsole registrieren. Nachteil, wir haben eine Ausgabe auf allen Konsolen.

Sniger87, 2016-03-28:
gleich reinschreiben finde ich keine gute idee, da wie du schon geschrieben hast dann alles auf jeder konsole ist
Ich debugge das morgen mal direkt auf dem PI und schau mal wo es da beim registrieren fehlt schlägt, wobei das ja anscheinden wieder nur sporadisch passiert

Symen, 2016-03-28:
Problem ist registriert wird es ja, sonst bekämen wir ja überhaupt keine Ausgabe auf der CLI-Konsole. Aber anscheinend greifen die Logging-Funktionen in den Child-Prozessen auf eine 'veraltete'-Version des Loggers zu. Anders kann ich mir das Verhalten nicht erklären.

Symen, 2016-03-28:
Wie lange soll eigentlich der Thread im MultiProcessingHandler laufen? Also nach meinem Verständnis müsste er eigentlich bis zum Ende laufen. Allerdings wird die Funktion 'receive' die der Thread aufruft nicht so lange ausgeführt wie erwartet sondern vorher beendet. Was glaub ich dazu führt, dass log-results die gesendet werden nicht ausgegeben werden.

PR #195 (2016-03-30): Bug cli log #188
/dev/console to /dev/tty1

Following tickets are finished:
resolved #188

tested on Pi:
- [x] test_R_Server_VLAN.py
- [x] test_AP_*.py
- [x] test_A_Server_2.py
closed 2016-03-30 by Symen